mcci-catena/arduino-boards #9: Permissions are wrong on STM32 tools on macOS, Linux

terrillmoore, 2018-03-21:
Up to and including 1.0.6 of the STM32 board package, the permissions in the release file were wrong when installing on macOS and Linux. It's possible that this is now fixed with the updated [MCCI board support repo](https://github.com/mcci-catena/Arduino_Core_STM32). But we should check 1.0.7, and if it's still broken, review permissions in the MCCI fork.

terrillmoore, 2018-03-23:
Confirmed that this is a problem with the Linux port. Builds work (so the compiler is all right) but permissions for the download tools are wrong.  

The quick workaround:

```shell
cd ~/.arduino15/packages/mcci/tools/STM32Tools/2017.9.22-mcci/tools/linux
find . -type f | xargs file | grep 'Bourne-Again' | cut -d: -f1 | xargs chmod +x
find . -type f | xargs file | grep 'ELF' | cut -d: -f1 | xargs chmod +x
```

sjpark-mcci, 2018-03-23:
Hi Terry,

 

I tried this and I am still getting an error:

 

Sketch uses 10456 bytes (5%) of program storage space. Maximum is 196608 bytes.

Global variables use 2512 bytes (12%) of dynamic memory, leaving 17968 bytes for local variables. Maximum is 20480 bytes.

/home/sjpark/.arduino15/packages/mcci/tools/STM32Tools/2017.9.22-mcci/tools/linux/stlink_upload: line 4: /home/sjpark/.arduino15/packages/mcci/tools/STM32Tools/2017.9.22-mcci/tools/linux/../linux64/stlink_upload: Permission denied

 

However, I changed the file format to tar.gz and tested on my linux VM. It is working fine now. Since the BSP package remains the same, I am  not sure how you want to proceed this. Do we need to wait for the next BSP update?

 

Thank you,

Sungjoon

sjpark-mcci, 2018-03-23:
Ah, the reason the workaround didn’t work is that you’re using linux64, and my workaround was for 32-bit linux.

 

But if the problem is resolved, then we know how to do this. Please test installing on Windows with the .tgz format.

 

--Terry

sjpark-mcci, 2018-03-23:
Also tested on Windows and it is working.

 

--Sungjoon

 

From: Terry Moore [mailto:tmm@mcci.com] 
Sent: Friday, March 23, 2018 2:56 PM
To: 'Sungjoon Park'; 'mcci-catena/arduino-boards'; 'mcci-catena/arduino-boards'
Cc: 'Assign'
Subject: RE: [mcci-catena/arduino-boards] Permissions are wrong on STM32 tools on macOS, Linux (#9)

 

Ah, the reason the workaround didn’t work is that you’re using linux64, and my workaround was for 32-bit linux.

 

But if the problem is resolved, then we know how to do this. Please test installing on Windows with the .tgz format.

 

--Terry

terrillmoore, 2018-03-23:
OK; moving this issue to https://github.com/mcci-catena/Arduino_Core_STM32/issues/16; we'll close this when we release 1.0.8.

terrillmoore, 2018-08-02:
Fixed in v1.0.8.

closed 2018-08-02 by terrillmoore