mcci-catena/arduino-boards #13: Serial1

mhmayyan, 2019-02-18:
I am trying to use the hardware serial port on Catena4612 (D0, D1). I used the SerialPassThrough example as shown below:-

`void setup() {
  Serial.begin(115200);
  Serial1.begin(115200);
}

void loop() {
  if (Serial.available()) {      // If anything comes in Serial (USB),
    Serial1.write(Serial.read());   // read it and send it out Serial1 (pins 0 & 1)
  }

  if (Serial1.available()) {     // If anything comes in Serial1 (pins 0 & 1)
    Serial.write(Serial1.read());   // read it and send it out Serial (USB)
  }
}`

The compilation output is shown in the attached file.

[SerialProblemOut.txt](https://github.com/mcci-catena/arduino-boards/files/2877729/SerialProblemOut.txt)

terrillmoore, 2019-02-20:
Sorry you're having problems!  This used to work, but I think we might have inadvertently taken this out in the recent BSP release. @chwon64, am I right? I don't see a USB + Hardware Serial setting on the 4612.

chwon64, 2019-02-20:
Yes, that's correct. There is no USB + Hardware Serial setting.
So it is expected link error -- undefined Serial1.

terrillmoore, 2019-02-20:
@mhmayyan -- I entered an issue in the BSP https://github.com/mcci-catena/Arduino_Core_STM32/issues/70.

The BSP is rather limited; it would be great if you could locally declare an object in your sketch by writing;

```c++
HardwareSerial Serial1(USART1);
```

But I don't think that will work because of how they handle events. We'll have to rewrite that portion because it's very limiting for us.

However, if you are brave and want to edit your platform.txt, you should be able to add `HAVE_HWSERIAL1` to the list of `-D` symbols at compile time. You'll see how it's manipulated by the menus.

mhmayyan, 2019-03-19:
**Thank you, Terry, for this help.
I actually got the Serial1 working by the following modifications:-**

In file: ../hardware/stm32/2.2.0/cores/arduino/HardwareSerial.cpp; line: 33
Before:                `#if` !defined(NO_HWSERIAL)`
After:                  `#if !defined(NO_HWSERIAL) || defined (FPRCE_USE_SERIAL1)`

In file: ../hardware/stm32/2.2.0/cores/arduino/HardwareSerial.cpp; function: HardwareSerial::HardwareSerial(void* peripheral)
Before:		`#if defined(Serial) && defined(PIN_SERIAL_RX) && defined(PIN_SERIAL_TX)`
After:		`#if defined(Serial) && defined(PIN_SERIAL_RX) && defined(PIN_SERIAL_TX) && !defined(FPRCE_USE_SERIAL1)`

In file: ../hardware/stm32/2.2.0/boards.txt; line: 455
Before:		`mcci_catena_4612.menu.xserial.usb.build.xSerial={build.usb_flags} -DNO_HWSERIAL` 
After:		`mcci_catena_4612.menu.xserial.usb.build.xSerial={build.usb_flags} -DNO_HWSERIAL -DFPRCE_USE_SERIAL`

**For testing purpose, I used the following:-
In my code; function: void setup(void), I added** 
`Serial1.begin(115200);`
**and in function: void loop(void), I added**
```
  while (Serial1.available()) {     // If anything comes in Serial1 (pins 0 & 1)
    Serial.write(Serial1.read());   // read it and send it out Serial (USB)
  }
```

**However, I don't know if these modifications are consistent and safe for the rest of the libraries and classes.**

svelmurugan92, 2019-09-05:
@mhmayyan Our latest BSP release has feature to enables both USB serial + HW serial. 
Please update the BSP and try.

mhmayyan, 2019-09-06:
@svelmurugan92 yes, it works. That is neat.
Thank you so much.

mhmayyan, 2019-09-08:
@svelmurugan92 With 2.5.0 and 2.4.0 I was unable to see the device on the computer. Serial CDC seems not working. With 2.3.0, however, it works and I can see the device listed on Tools>Port.

I used the example sketch Communication>ASCIITable

closed 2019-09-12 by mhmayyan