mcci-catena/arduino-boards #17: Release v2.6.0 of STM32 and v2.1.0 of SAMD

terrillmoore, 2019-10-29:
Please release [STM32 v2.6.0](https://github.com/mcci-catena/Arduino_Core_STM32/releases/tag/v2.6.0) and [SAMD v2.1.0](https://github.com/mcci-catena/ArduinoCore-samd/releases/tag/v2.1.0). Remember that we'll have to clear most CI caches.

kent-williams, 2019-10-31:
@terrillmoore We are hoping to release our examples targeting your board in the next day or two! We have instructions right now on how to add the library ldflags bit to the board support package, but would love to remove that if this gets released in time! Thanks again for the help!

terrillmoore, 2019-10-31:
@kent-williams It's released already, I think! See https://github.com/mcci-catena/arduino-boards/commit/d561b11320eb2dc8dc1f01882469f6a876221a8c I believe @sjpark-mcci just didn't close the ticket yet. (Am I right, Sungjoon?)  It would be clearer if we also tagged this repo, but since it has multiple things with different versions, versioning is tricky.  Sungjoon, why not tag this repo as v2.0.0 (there are breaking changes for CI that I"m having to fix)?

kent-williams, 2019-10-31:
My apologies! My arduino IDE wasn't downloading the latest json! Thanks again!

terrillmoore, 2019-10-31:
No problem at all, happy to help out. Thanks for your interest!

closed 2019-10-31 by terrillmoore

sjpark-mcci, 2019-10-31:
Sorry I did not close the ticket! Yes it is released already and I will tag this repo as v2.0.0.